microsoft/Windows-iotcore-samples #76: It's hard to accept incoming bluetooth pairing requests

sergeyDoroshenko, 2018-01-26:
"IoT Core Default App" uses the same ObservableCollection instance for keeping both incoming BT pairing requests and the BT devices located around. Every update of a BT device nearby triggers an update of the ObservableCollection instance. Updates of the ObservableCollection object remove all incoming BT pairing requests.  When there are a few BT devices around, the ObservableCollection content changes too frequently. So it's getting very complicated/impossible to accept any incoming BT requests.

sergeyDoroshenko, 2018-01-30:
@namkedia

saraclay, 2018-01-31:
Hi @sergeyDoroshenko, have you tried the latest IoT builds? The code was updated roughly two weeks ago. This should have helped this issue.

saraclay, 2018-01-31:
As an FYI - the UI will look different with separate inbound and outbound pairing information status lines.

sergeyDoroshenko, 2018-02-01:
Thanks @saraclay. Do you have ETA for new code on GitHub?

saraclay, 2018-02-07:
Hi @sergeyDoroshenko, the Insider Build 17083 is available. See release notes here: https://docs.microsoft.com/en-us/windows/iot-core/release-notes/insider/17083

closed 2018-02-14 by saraclay