microsoft/Windows-iotcore-samples #92: Why does the link of opencv documentation is redirected and cannot be seen now

Ichangeling, 2018-02-14:
The documentation of opencv sample is redirected and cannot be seen from last week.
Is there any way to see that now?
@namkedia

saraclay, 2018-02-14:
Hello - can you please repro what you're seeing?

Ichangeling, 2018-02-14:
[image]

Ichangeling, 2018-02-14:
https://github.com/Microsoft/Windows-iotcore-samples/blob/develop/Samples/OpenCVExample/CPP/README.md

If I clicked the link of the documentation on this page, then instead of the doc, it shows

> Redirecting you to WindowsOnDevices.com... Click here if you are not automatically redirected.

And then I just redirected back to the previous page again.

saraclay, 2018-02-21:
Ah, thanks for pointing this out. I will get this fixed.

saraclay, 2018-02-23:
Hello - this problem has been resolved. Thank you for pointing this out!

closed 2018-02-23 by saraclay