microsoft/Windows-iotcore-samples #148: HID injection sample and article missing

wind-rider, 2018-05-21:
Previously there were a sample and article on HID injection using the virtual hid framework, but they have either not been migrated or have been removed.

What is the reason for their removal? Could you please update them and put them back online?

saraclay, 2018-05-22:
Hi @wind-rider - this sample was implemented to work around the lack of input injection but was not getting a lot of traction. What are you looking to use it for?

wind-rider, 2018-05-23:
Hi @saraclay, the bigger picture I’d like to solve is to [listen to BLE GATT messages from a Surface Dial-like device and inject them as HID reports](https://stackoverflow.com/q/45773666/1114918), so that I can use that device with apps that support IRadialController. You can find the homework I did documented at that link.

One of your colleagues answered there but he gave a (partial) answer without explaining his reasoning and reflecting much upon my research, so bottom line it was still difficult for me to learn from it. Any further clarification to that [question on Stackoverflow](https://stackoverflow.com/q/45773666/1114918) would still be greatly appreciated - maybe you know of a colleague who can assist further?

W.r.t. the sample in particular - in my memory it was a bit "cleaner" and better explained than the [virtual hid sample in the Windows Driver Samples repository](https://github.com/Microsoft/Windows-driver-samples/tree/master/hid/vhidmini2). I referred to it from my question on stackoverflow and would like to read it again.

saraclay, 2018-06-07:
Hi @wind-rider, let me see what I can do but if you can forge on, I would recommend that so we don't act as a blocker. Thank you.

closed 2018-06-07 by saraclay

hansmbakker, 2018-09-02:
@saraclay have you found anything that could help me?