microsoft/Windows-iotcore-samples #150: SerialUART:   SerialDevice.FromIdAsync() Exception on Pi3 Internal Broadcom UART

PRIMETSS, 2018-05-25:
The SerialUART sample seems to fail on a R-Pi3 using internal on board UART

The internal UART that comes with the Rasberry PI3 which is Enumerated as device ID '\\?\ACPI#BCM2836#0#{86e0d1e0-8089-11d0-9ce4-08003e301f73}' using DeviceInformation.FindAllAsync()
Causes an exception when atempting to get the SerialPort via SerialDevice.FromIdAsync()

> Value does not fall within the expected range.

Spent a few days trying to get this going, and then found adding this to the manifest fixed the issue.

> <DeviceCapability Name="serialcommunication">
      <Device Id="any">
        <Function Type="name:serialPort" />
      </Device>
    </DeviceCapability>

It did work for about a week, and then, as per other reports of people also suggesting it was 'intermittent' I now not being able to get it working again after a further few days.

After a bit more research I found this artical that suggest on-chassis UART are not supported
https://docs.microsoft.com/en-us/uwp/api/Windows.Devices.SerialCommunication

And that they should be filtered out of FindAllAsync() enumerations.

My question is, what is the reason for this? Is it an issue with the handling of the onboard Broadcom UART? Is there plans to fix this? As after spending alot of time on this, seems a waste that the on-board UART cant be used!

Thanks!

jordanrh1, 2018-06-11:
What build are you running?

    Reg query "HKEY_LOCAL_MACHINE\SOFTWARE\Microsoft\Windows NT\CurrentVersion" /v BuildLabEx

We are seeing an issue in latest insider preview builds where internal UARTs cannot be opened with the SerialCommunication class.

PRIMETSS, 2018-06-11:
BuildLabEx    REG_SZ    17686.1003.armfre.rs_prerelease.180603-1447

jordanrh1, 2018-06-12:
This build is affected by the known issue. We're working on a fix. Thanks for your patience.

hnky, 2018-06-26:
Is this build also affected?
BuildLabEx    REG_SZ    17134.1.amd64fre.rs4_release.180410-1804

PRIMETSS, 2018-06-26:
Wondering if it got fixed in 
Release Notes for Build 17692?

saraclay, 2018-06-27:
@hnky - I don't think the latest commercial release is affected.
@PRIMETSS - The fix is in. Please try this with build 17692. Let us know if you have issues.

closed 2018-06-27 by saraclay

PRIMETSS, 2018-07-03:
Just tried with 17692 and updated to 10.0.17704.1000
and dont think this is fixed. Same issue with finding UART device ID "\\\\?\\ACPI#BCM2836#0#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"

But SerialDevice.FromIdAsync(serialDeviceFound.Id) // ID above

still has exception 

"Value does not fall within the expected range."