microsoft/Windows-iotcore-samples #158: using System; not recognized

sdewailly, 2018-06-12:
Hi,

After loading your project on VS, the IDE doesn't seem to recognize the line: using System;
After redading google results, it seems that this namespace belongs to the .NET Framework.

So, do I have to update the .NET Framework on my computer even if it is preinstalled on Windows 10?

thx

bfjelds, 2018-06-14:
Which sample are you seeing this for?

What version of Visual Studio are you using?  Do you see the same issue when creating a new UWP app?  If you see the same issue when creating a new UWP app, you may have an issue with your Visual Studio installation ... if you do not see this issue, what version of .NET is being used for the new project?

saraclay, 2018-07-03:
Closing due to lack of response.

closed 2018-07-03 by saraclay