microsoft/Windows-iotcore-samples #170: SerialUART: SerialDevice.FromIdAsync() Exception on Pi3 Internal Broadcom UART #150

PRIMETSS, 2018-07-03:
Just tried with 17692 and updated to 10.0.17704.1000
and dont think this is fixed. Same issue with finding UART device ID "\\?\ACPI#BCM2836#0#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"

But SerialDevice.FromIdAsync(serialDeviceFound.Id) // ID above

still has exception

"Value does not fall within the expected range."

still have below in manifest

saraclay, 2018-07-20:
Closing for now. Please update if this still is not fixed.

closed 2018-07-20 by saraclay

PRIMETSS, 2018-07-20:
It wasn't fixed. Was monitoring this after it got assigned. But don't think there was any update.
No new build since I.P . 10.0.17704.1000

jordanrh1, 2018-08-09:
Sorry for the confusion. The initial fix is available as of build 18217, but the fix has issues. We are continuing to work with the team that owns the capability access manager.