microsoft/Windows-iotcore-samples #200: Windows IoT Extensions for the UWP not available

Informaticore, 2018-08-31:
I try to do the [HelloWorld](https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/HelloWorld/CS) sample but while setting up Visual Studion and the project, I am not able to find or install the Windows IoT Extensions for the UWP.
In the list where I can add references I can only find the Windows moble Extensions for the UWP but not for IoT.

PRIMETSS, 2018-08-31:
Have you installed the Windows 10 SDK (Preview or matching your version?)
Latest Preview here if your an insider
https://www.microsoft.com/en-us/software-download/windowsinsiderpreviewSDK

Informaticore, 2018-08-31:
> Have you installed the Windows 10 SDK (Preview or matching your version?)
Yes! And all the other as well even for Windows 8.1 (but I deleted them after it was not helping)

Do I need to use the insider preview? Or should it work with the normal one as well? (I am not an insider)

PRIMETSS, 2018-08-31:
Get the SDK that matches the version you have installed on the Pi
Your dev machine might need to be same or higher version

Informaticore, 2018-08-31:
I have the matching SDK on both. On the Pi the version number is the same as the on installed on my dev machine! The extension is still not in the list of references. 
I checked a few tutorials, I uninstalled the whole Visual Studio and installed it again and installed it like explained - but the extension never appeared, only the one for mobile.

I am using the community Edition - are there any known limitations?

Informaticore, 2018-08-31:
The folder 'C:\Program Files (x86)\Windows Kits\10\Extension SDKs\' only contains 'WindowsMobile'
Raspberry Pi Windows IoT Version: v.10.0.17134.1

In the Installer List it is .1 at the end butthere is no .1 to download so I would guess, thats okay?!
Here is the list from the Installer of alle installed tools:
[image]

What else is missing?

paulmon, 2018-08-31:
The extension SDK doesn't exist for SDKs with a version greater than 17134.  The APIs that were in the extension SDK have been merged into the main UWP SDK.  If you are using an SDK with a version greater than 17134 you should delete the reference to the extension SDK

PRIMETSS, 2018-09-01:
Add Reference / Extention
[image]

Informaticore, 2018-09-01:
@paulmon does that mean I do not need that extension anymore?? (just to make sure)

Informaticore, 2018-09-01:
It is working now without any extension - that should be mentioned in the tutorials!
@PRIMETSS nice hint but as I said, this list item does not exist!

closed 2018-09-01 by Informaticore

PRIMETSS, 2018-09-01:
hmmm Odd as for version 17134 it should be there.

As Paul said, for version above the API's where merged into the main SDK so no specific extension no longer needs to be referenced.
I guess if your up and running then that's good.
And yes understand the bit of frustration re- the Docs....
But stick with it, as its ALMOST impossible to keep the Docs (found via google) up to date with a moving platform.... That's a Nobel Peace Prize achievement in my books!

Informaticore, 2018-09-01:
@PRIMETSS I see but at least the installation should be clear. I mean I am doing development for years now but I can pin this experience tp my top 3 shittiest :D 
Now it is running - time to do some fancy stuff.

sunnysonydas, 2019-03-22:
I try to do the HelloWorld sample but while setting up Visual Studion and the project, I am not able to find or install the Windows IoT Extensions for the UWP.
In the list where I can add references I can only find the Windows moble Extensions for the UWP but not for IoT.

myokeeh, 2019-03-24:
It is no longer required. It's been "merged" with the main Windows 10 SDK. Some of the older documentation need updating.

sunnysonydas, 2019-04-11:
I am facing problems with my Visual Studio when i am trying to create a IoT app This is --

 "visual studio requires a newer version of windows to display the content. please update to windows 10 version 1809 (10.0.17763.0) or later"

Now how can I get rid  from this problem? ( Note: My Windows OS is activated and updated)

paulmon, 2019-04-12:
Run winver.exe to see what version of Windows you have installed.  The build number must be less than or equal to the target version of your project.  Alternatively if you are not using any new APIs you can select a target version that is less than or equal to the version of Windows you have installed.

This link may be helpful: [Choose a UWP Version](https://docs.microsoft.com/en-us/windows/uwp/updates-and-versions/choose-a-uwp-version)

endyb66, 2019-04-28:
There is no Windows IoT Core Extension fo version 17763, when targeting build 1809. Is this missing only inside my VS installation?

myokeeh, 2019-04-28:
I repeat: It's been "merged" with the main Windows 10 SDK. You don't need to reference it anymore.

endyb66, 2019-04-29:
So when upgrading my project to 1809, i can remove the extension from my references without any problems?

myokeeh, 2019-04-29:
That's my experience with 17763 already.

endyb66, 2019-04-29:
Ok perfect, thank you

steam3d, 2020-01-17:
But i have error "The name 'FullTrustProcessLauncher' does not exist in the current context UWP"
I use 17763 and can't add extension. What's wrong?

Zeyneb-Rh, 2020-07-13:
@paulmon Please I have a problem I am beginner at programming, I hope u can help me;I have installed on my SD card IoT v.10.0.17763.107, on RPi 3, and It boots good. I can also see the device on Dashboard, and Device portal, working from Windows 10 Pro Version 18363.900. However the following issues are observed.

Visual Studio 2019, Universal application, identifies my remote system, but rejects to deploy in following configurations
a) ARM with Universal (Unencrypted Protocol),
Severity Code Description Project File Line Suppression State
Error DEP6957: Failed to connect to device '192.168.1.3' using Universal Authentication. Please verify the correct remote authentication mode is specified in the project debug settings. COMException - Exception from HRESULT: 0x801C0003 [0x801C0003] 
b) ARM - Windows/None mode with following error
Error: Unable to connect to the Microsoft Visual Studio Remote Debugger . The Visual Studio 2019 Remote Debugger (MSVSMON.EXE) does not appear to be running on the remote computer. This may be because a firewall is preventing communication to the remote computer. Please see Help for assistance on configuring remote debugging.

I don t know how to resolve this problem. I will be  so thankfull.

paulmon, 2020-07-13:
Usually restarting Visual Studio will fix this error if you are building a UWP application.

If you building device drivers see this page:
https://docs.microsoft.com/en-us/windows/iot-core/learn-about-hardware/driverdeployment

If you are developing a UWP application see link:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/buildingappsforiotcore

If you are developing a console app see this page:
https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/remotedebugging