microsoft/Windows-iotcore-samples #231: When can we get source code of IoTDefaultApp 3.x? And there's a bug report.

sunwayking, 2018-10-22:
1. Please upload 3.x source code of IoTDefaultApp.
2. Try to analyse this Language setting bug from source code:
After setting language to Chinese(Simplified) in IotDefaultApplication's System setting page, Navigate to other page(in Hamburger Menu) and come back,  the settings will be lost and language combo box displays nothing. What's more, With setting system language to Chinese(Simplified), Multilingual applications including Simplified Chinese language still runs like an en-us application, no zh-CN words show.
PS：IotDefaultApplication's version is 3.1.0.0

saraclay, 2018-10-23:
Hi @sunwayking, I will send this feedback to the team. Please stay tuned.

saraclay, 2018-11-12:
Hi @mawett - we are getting this pushed through today! Also, could you please email me at saclayt@microsoft.com for more information on what you're doing? We'd love to hear about it.

saraclay, 2018-11-12:
The new code can be found here:
https://github.com/Microsoft/Windows-iotcore-samples/tree/master/Samples/IoTCoreDefaultApp

Thank you for your patience!

closed 2018-11-12 by saraclay