microsoft/Windows-iotcore-samples #233: Broken link in README.md

kn-neelalohitha, 2018-10-26:
# What happened?
I was going through [README.md](https://github.com/Microsoft/Windows-iotcore-samples/blob/develop/README.md) as I was interested to learn about Windows 10 IoT Core and use it for my project.When i clicked on "Gett Started" it redirected to [Error 404 - "Sorry, we couldn't find that page."](https://developer.microsoft.com/en-us/windows/iot/getstarted)

# What you expect to happen?
I expected to be taken to "Get Stared" guide of Windows 10 IoT Core.

# Reference:
In [README.md](https://github.com/Microsoft/Windows-iotcore-samples/blob/develop/README.md) under the first paragraph of **Windows 10 IoT Core samples** in the **Note** section go to line 5 : **Get Stared**.

kn-neelalohitha, 2018-10-26:
Raised PR : [#234](https://github.com/Microsoft/Windows-iotcore-samples/pull/234).
Replaced the broken link in README.md with the "**Get Started**" link in [Windows 10 IoT documentation](https://docs.microsoft.com/en-us/windows/iot-core/).
Please review.

saraclay, 2018-11-12:
Done! Thanks so much, @kn-neelalohitha!

closed 2018-11-12 by saraclay

kn-neelalohitha, 2018-11-13:
@saraclay You are welcome and I am always glad to contribute!