microsoft/Windows-iotcore-samples #235: "Access is denied" when scanning for wifi networks in IoT Onboarding solution.

CameronAavik, 2018-10-30:
I downloaded the [IoT WiFi Onboarding](https://github.com/Microsoft/Windows-iotcore-samples/blob/develop/Samples/IoTOnboarding/CS) solution and went through all the steps in the README. I was able to successfully follow every step until Step 4.8 in which I encountered an issue.

When the `OnboardingService` attempts to call `WifiAdapter.ScanAsync` ([see line](https://github.com/Microsoft/Windows-iotcore-samples/blob/bde9f809540373a751e35c3e3819c5f9b4dcc022/Samples/IoTOnboarding/CS/IoTOnboardingService/OnboardingService.cs#L676)), I get the following error:

    Exception thrown: 'System.UnauthorizedAccessException' in System.Private.CoreLib.ni.dll
    Access is denied. (Exception from HRESULT: 0x80070005 (E_ACCESSDENIED))

The stack trace is just:

    at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess(Task task)

The reason I downloaded this solution is because I was getting the same error when implementing this on my own IoT Core application and this project claimed that it would get it working with a headless application. My belief is that this was never meant to work according to the [docs for ScanAsync](https://docs.microsoft.com/en-us/uwp/api/windows.devices.wifi.wifiadapter.scanasync) which claim that:

>You should call this API from a foreground app. If it is called from a background task, you will see an "Access Denied" error message or a similar message.

The [Microsoft docs](https://docs.microsoft.com/en-us/windows/iot-core/connect-your-device/setupwifi) for Wi-Fi onboarding list this sample as a way for connecting to a Wi-Fi network from a headless application, so I'm not sure how this sample is supposed to get around this limitation.

Information:
- Windows v.10.0.17134.285
- Raspberry Pi 3

saraclay, 2018-11-12:
Hi @CameronAavik, thank you for the thorough repro instructions. I'm discussing with our team about this.

saraclay, 2018-11-19:
Hi @CameronAavik and @tbuquet, I have created an issue for our team. However, in order to triage it properly, I will need to know if this is a blocking issue for either of you. Thank you.

CameronAavik, 2018-11-19:
Yes, this is a blocking issue for me.

saraclay, 2018-11-26:
Hi @CameronAavik and @tbuquet, I apologize for the delay. You might get a quicker response from our MSDN forums as this is less of a doc issue and more of a technical issue: https://social.msdn.microsoft.com/forums/en-US/home?forum=WindowsIoT

saraclay, 2018-11-26:
Please feel free to submit a change to our docs when the issue is solved in the forums.

closed 2018-11-26 by saraclay