microsoft/Windows-iotcore-samples #283: turn off monitor

honorew, 2019-01-14:
Hi,is there any way to turn on/off monitor on iotcore?thanks

saraclay, 2019-01-14:
@honorew can you be more specific? Are you referring to the IoT Core Default App?

honorew, 2019-01-15:
I want to realize function that can turn on/off monitor like sleep mode on windows desktop.

PRIMETSS, 2019-01-15:
Which type of display?
If its a a I2C display this might help

https://lb.raspberrypi.org/forums/viewtopic.php?t=152495
https://www.raspberrypi.org/forums/viewtopic.php?f=105&t=152495

saraclay, 2019-01-22:
Closing this out for now - hoping the links to the forums help. If not, feel free to reopen/create a new issue.

closed 2019-01-22 by saraclay

honorew, 2019-01-23:
oh it dosen't work for me.

PRIMETSS, 2019-01-23:
Just tried it out for you to see if I could get it working, but sadly I only have a 3B+ (free at moment) and of course the TouchScreen driver is disabled and no new Updates for this version for along time!! (Sadly!). Had a bit of a look at trying to enable it, but no luck.
I'll try it on another older board when free.

honorew, 2019-01-24:
Thank you for your attention.My screen is just a normal screen(AOC).

PRIMETSS, 2019-01-24:
OHHH a HDMI screen?

Hmm just tried this for you after a bit of research, calling a Win Interop API
Works on my desktop, but looks like API calls don't work (naturally) on Win10IoT
Suprised it didnt cause an exception or write anything to the Windows Event log... 

[https://github.com/PRIMETSS/Win10IotDisplayOff](url)

honorew, 2019-01-24:
yeah  I've tried this before, thanks.