microsoft/Windows-iotcore-samples #334: IoT Core project templates for VS2019

hansmbakker, 2019-03-03:
Will you make the [IoT Core project templates](https://github.com/ms-iot/samples/tree/develop/IotTemplates/WindowsIotCoreTemplates) available for VS2019?

I'm asking the question here since the original repo is archived.

saraclay, 2019-03-12:
@hansmbakker I'm checking with the team. Stay tuned.

saraclay, 2019-03-13:
Yes, they are available for VS2019!

paulmon, 2019-03-13:
There is a new repo also, that is not archived: https://github.com/ms-iot/vs-project-templates/tree/master/WindowsIotCoreTemplates

and the Visual Studio marketplace page is here: https://marketplace.visualstudio.com/items?itemName=MicrosoftIoT.WindowsIoTCoreProjectTemplatesforVS15

hansmbakker, 2019-03-13:
Thank you! Then this can be closed

closed 2019-03-13 by hansmbakker