microsoft/Windows-iotcore-samples #342: SerialUART sample not working on Windows 10 Pro

luyaok, 2019-04-01:
There is a customer report that SerialUART sample not working on Windows 10 Pro (Version    10.0.17134). SerialDevice.FromIdAsync(dis[0].Id) always returns null. He use the following device:

`dis[0].Id = "\\\\?\\FTDIBUS#VID_0403+PID_6001+A906I45PA#0000#{86e0d1e0-8089-11d0-9ce4-08003e301f73}"`

But he get it working on Raspberry Pi 3B+.

So is this issue related to Windows edition Pro?

MSDN thread: https://social.msdn.microsoft.com/Forums/en-US/980892e7-2b29-4c30-904b-9d885b74c9d7/serialport-await-serialdevicefromidasyncdis0id-returns-null?forum=WindowsIoT

saraclay, 2019-04-03:
Hi @luyaok, have you tried [embedded mode](https://docs.microsoft.com/en-us/windows/iot-core/develop-your-app/EmbeddedMode)? It's on both IoT Core and IoT Enterprise, though I'm not sure about Windows 10 Pro as that is not our domain.

luyaok, 2019-04-08:
The customer got it working now. 
https://github.com/Microsoft/Windows-iotcore-samples/blob/develop/Samples/IoTCoreDefaultApp/CS/SmartDisplay/Presenters/ConnectedDevicePresenter.cs
So I close this issue.

luyaok, 2019-04-08:
Solved.

closed 2019-04-08 by luyaok