microsoft/Windows-iotcore-samples #350: Why can't use the diagnostic tools after combine two projects?

moyanming, 2019-04-08:
Hi, 
According to the steps [here](https://github.com/Microsoft/Windows-iotcore-samples/tree/develop/Samples/ForegroundAppWithBackgroundApp/CS), all works fine. But why the diagnostic tools can't be used?

saraclay, 2019-04-08:
@moyanming where does it say that the tools can't be used when debugging? Have you tried it?

moyanming, 2019-04-09:
Hi @saraclay ,
Yes, I have test this combine solution but there is no diagnostic when debugging.
[image]

paulmon, 2019-04-11:
The diagnostic tools do not work with background applications.  Could that be your problem?

moyanming, 2019-04-12:
Hi @paulmon ,
17763.253, DB410C
The "MyForegroundApp" is the startup project. Why there is no foreground app but only have the background app after deploy this solution?
[image]

Which type of this solution is, foreground or background?
How to debug the foreground one (MyForegroundApp) as there is no foreground app was deployed?

moyanming, 2019-04-12:
BTW, please note that there is another two same name app, headed and headless are all have the same name "MyTestWDPHeadless-uwp_...". This project is a background app but I added a BlankApp (foreground) into this solustion refers [here](https://docs.microsoft.com/en-us/windows-hardware/service/iot/servicing-msstore#special-instructions-for-headless-apps).
What is the different between solutions of combine one and this project (MyTestWDPHeadless)?
[image]

paulmon, 2019-04-12:
> 1. Why there is no foreground app but only have the background app in the WDP after deployed this solution?
This is a known issue.  The foreground app should be displayed but it isn't

> 2. Why the name of the backgournd app is "MyForegroundApp" but not "MyBackgroundApplication"?
>    [image]
The name comes from the appxmanifest and is the package name not the Visual Studio project name.

> 3. Which type of this solution is, foreground or background?
It is both.

> 4. How to debug the foreground one (MyForegroundApp) as there is no foreground app in the WDP?
Select the foreground app as the startup app in Visual Studio and press F5

> 5. There are two same name apps of this solution (usb "iotstartup list") after deployed: one is headed and another one is headless. This is confused.
This is by design.  The format is the <package name>!<Application>.  !App identifies which <Application> in the appxmanifest is being identified.  Headless means that the thing identified is a BackgroundTask of type "startup"

paulmon, 2019-04-12:
> 6. What is the different between solutions of combine one and this project (MyTestWDPHeadless)?

The solution [here](https://docs.microsoft.com/en-us/windows-hardware/service/iot/servicing-msstore#special-instructions-for-headless-apps) creates a backgroundApp with stubbed foreground app that does nothing.  A foreground app is required to publish a background app to the store, even if the foreground does nothing.  As far as I know the scenario where the foreground app does nothing is only allowed on Windows IoT Core.

closed 2019-04-23 by saraclay