microsoft/Windows-iotcore-samples #464: ShellLauncher V2 script issue

MaterialStep4, 2020-06-15:
I am attempting to use the powershell WMI bridge helpers and the provided xml script in Windows 10 b1909.  
I have tried various ways of escaping the unescaping the xml.  The result is the same. 

My command, run from PS in system context:
PS C:\Windows\system32> Set-ShellLauncherBridgeWMI -Filepath .\ShellLauncherSid2.xml

The bridge script throws this message:

=========================================================
Set-CimInstance : A general error occurred that is not covered by a more specific error code.
At C:\Windows\system32\ShellLauncherBridgeWmiHelpers.ps1:17 char:5
+     Set-CimInstance -CimInstance $AssignedAccessCsp
+     ~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
    + CategoryInfo          : NotSpecified: (MDM_AssignedAcc..../Vendor/MSFT"):CimInstance) [Set-CimInstance], CimExce
   ption
    + FullyQualifiedErrorId : MI RESULT 1,Microsoft.Management.Infrastructure.CimCmdlets.SetCimInstanceCommand
======================================================

MaterialStep4, 2020-06-15:
I also got this error on a patched Windows 10 1809 build.

ssrakesh, 2021-06-21:
Sorry, This replay almost an year late.
Main reason for the error message, is due to either syntax in xml or error in xml format itself.
Did u check the xml for any errors? any missing tags?.
Edit the xml in Visual studio and fix errors.